VoronDesign/Voron-2 #139: Too long mounting screws for RS-25-5 in V2.2 manual

jacksterz, 2020-05-20:
The V2.2 manual, at least on https://www.dropbox.com/s/96b65b5ddmter8s/VORON2.2%20Manual.pdf?dl=0 , is still recommending M3x8 SHCS for mounting the Meanwell RS-25-5 to the printed bracket **meanwell_rs_25_##_mounting_bracket_x1_Rev1.png**. The M3x8 are too long and they will chew through the insulation and make contact with the PCB of the PSU, possibly creating a fire hazard. 

It is a known issue which can be found in the Discord announcements, but the manual still has not been updated even though this announcement was made last November:
_"SERVICE BULLETIN: Please use M3x6 BHCS when mounting RS-25-5 PSU. M3x8 may go too far into the PSU body and puncture the insulating sheet, shorting out the main PCB. Manual will be updated shortly"_

jacksterz, 2020-05-20:
Update: I opened my PSU and noticed that one of the screws had indeed penetrated through the insulation and made contact with the PCB, but Meanwell has marked the mounting screw areas with black circles on the PCB and left them unpopulated. Therefore there should not be any actualy risk of a short.

FHeilmann, 2020-12-09:
V2.2 is an outdated design at this point. Unfortunately we no longer have the sources for the manual so we can't fix this issue. Furthermore it's hard to say how up to date the version you pulled from dropbox is.

Sorry for the incovenience

-Florian

closed 2020-12-09 by FHeilmann