VoronDesign/Voron-2 #141: Parts are loose

yhaiovyi, 2020-05-24:
There's 0.1 mm expansion compensation put directly into cad files. Now all parts I have already printed on a calibrated machine go into trash.

yhaiovyi, 2020-05-24:
A few examples:
- Bondtech axial bearing is 8mm. The corresponding hole size in CAD is 8.2mm.
- Z drive bearing is 20mm. The corresponding arcs sizes is 20.3mm!
Moreover it seems like compensation is not consistent, and meant to fix issues with some particular printer/filament.

yhaiovyi, 2020-05-24:
Now my own experience.  I never worked with materials that would give that bad expansion. the biggest expansion compensation I've ever done was 0.09mm, and most certainly not 0.15mm. 0.15mm is severe over extrusion.

Maglin, 2020-06-21:
.2-.3mm is standard ABS compensation for ABS shrinkage.  Some parts are designed for a sliding fit and not a press fit that would end up breaking the part from the stresses.

You never said what material you are using but every functional part of the Voron 2 printer is designed for ABS.  

Closing as this is as designed.

closed 2020-06-21 by Maglin