VoronDesign/Voron-2 #144: TEST_PRINT

prg3, 2020-06-12:


prg3, 2020-06-12:
The test prints buried inside of the Voron-2 repository is somewhat not intuitive for someone looking for test prints.. These would be better served by being in a repo of their own: VoronDesign/Voron-Tests or VoronDesign/Test-Prints

Maglin, 2020-06-21:
This has to do with how releases are handled.  The test prints are easily accessible for everyone that downloads the release as intended.

The test prints are only to verify your current printer is capable of printing parts that will work to build the Voron version you downloaded.  And then to help tune after you finish your build.  If we put them in another repo then the opposite of what you intend would happen.  Many people wouldn't know where they are located, or they wouldn't get added to a release on accident or worse yet.  We have two places for the same files and they end up out of sync with each other as they get revised.

prg3, 2020-06-21:
Right, but this is one place for all Voron, v0, v1 and v2 already, but they are in the buried in the v2 repo. It would make a lot of sense for these to be at one common location with a pointer to that location for all Voron releases.

And then use Git submodules as part of the build process to include test prints in every release.

Aggebitter, 2020-12-12:
The test prints should have there printed size and measurements included. Event ABS differs in how much it shrinks. Then it's easy to calibrate the print for the needed precision.

FHeilmann, 2020-12-14:
Using git submodules adds unnecessary complexity. 

As for your suggestion @Aggebitter. We'd welcome a PR submitted to VoronDesign/VoronUsers with improved test prints. If they prove to be universally better, we might include them in a future release.

I hope that answers your questions.

Bests,
-Florian

closed 2020-12-14 by FHeilmann