VoronDesign/Voron-2 #147: Voron VFA's how to remove them (small issue)?

homepods, 2020-06-20:
As for VFAs, one that is noticeable in every Voron 3D print, is some sort of Z banding, as the surface going up the bed, isnt perfectly straight and there is some sort of miss alignment of the extrusion. 
I am comparing Voron (because of it's price point) to everything above other printer in terms of quality.

Now, please don't ignore this issue, let;s find a way to fix it through cooperation.

I am trying to understand why this happens. Is it caused by Z guide rails, or AB rails? or E rails? Is it by the belt tension or by stepper motors?

It is caused by pulleys or by the whole design because guide tolerances  are variable?

Who agrees to have small Z-banding and wants to remove it?

doowrednu, 2020-06-21:
I don't have any z-banding - so not sure what you mean

arch-xanin, 2020-06-21:
VFAs and z-banding are two different things.  

I do not agree to have any z-banding, small or otherwise.

Tell us more about your voron and what you are seeing in your prints specifically please?

homepods, 2020-06-21:
2.4 it is not just me - I can see all pictures from voron showcase on discord to have layer miss-alignment going up the Z like Z-Banding. 

Also Nero made a live-stream and his cube had also z-banding.
https://www.youtube.com/watch?v=bURkB-ecNbw&t=14037s

Guys from Poland Druk3D have Z-banding.
https://youtu.be/X2vB1MGsWLk?t=890

It appears that not just me, but every print has some sort of Z banding.

Is this a mechanical or electronic derived miss-alignment. What do you guys think? Where should I focus my correction mod to fix the problem. 

I am trying to understand why this happens. Is it caused by Z guide rails, or AB guide rails? or E guide rails? Is it by the belt tension or by stepper motors?

It is caused by pulleys or by the whole design because guide tolerances are variable?

Maglin, 2020-06-21:
This is an effect of how 3D printed parts are made with FDM.  It's superficial and moving to a very low layer height resolution can make it seem to disappear.  This is a non-issue.  It's able to be seen in EVERY FDM print for any printer.  

My analogy for this would be similar to "why is the block of steel so heavy?" or "how come every time I put fire on this dry wood it burns?".  You are extruding a circular cross-section and when you stack them you will have this effect.  Of course, you wouldn't see it on say SLA, SLS, MJF etc because it's a process that leads to actual flat walls.

doowrednu, 2020-06-21:
If people do not tune their printers that is down to them - I see 1,000's of trash prints on an ender to every 1 decent one - it is the same for any 3d printer

homepods, 2020-06-21:
> This is an effect of how 3D printed parts are made with FDM. It's superficial and moving to a very low layer height resolution can make it seem to disappear. This is a non-issue. It's able to be [...]

Not every 3D printer has this issue. Some 3d printers, like the cheap ender 3, creality or others in that category. Yes, if you tune them as @doowrednu  says, can increase their quality, meaning to change their extruder, tighten their v-slot rails, change the termistor, the MCU and so on.

So my question is where should I look to tune it. Anyone has achieved zero Z-banding on a Voron, can I see some examples so I can build my confidence, cuz at the moment I am confused!!

VoronDesign, 2020-06-21:
This discussion is going nowhere. Please do some basic modicum of research.

closed 2020-06-21 by VoronDesign

homepods, 2020-06-21:
are you going to hide the issue instead of being open to look for a fix! lol ... it's a problem and there has to be a solution! don't ignore it cuz it will come to you sooner or later.

doowrednu, 2020-06-21:
This issue tracker is for owners of the printer - when you have built one and tuned it please feel free to create an issue. If you want to continue to discuss hypothetical problems - please do it elsewhere.