VoronDesign/Voron-2 #148: Readme for v2.1.1 has broken link

kauedg, 2020-06-20:
The README.md file in the v2.1.1 tag has a broken link to the Voron 2.1.1 configurator.

`The second official release of the Voron 2.1.1 3d printer. You can find the BOM in the configurator located at the [Voron Design](http://vorondesign.com/voron2.1) website.`

Either the link should me removed or the configurator should be made available.

kauedg, 2020-07-03:
Is it possible to make the configurator available again or it's removed because the version is considered obsolete?

FHeilmann, 2020-12-09:
V2.1 has been superseded twice at this point, so in order to keep clutter low and make sure people grab the correct version we chose to remove the parts related to the old printers.

At this point we wouldn't be able to guarantee their correctness either way and the branches are merely preserved for posterity.

-Florian

closed 2020-12-09 by FHeilmann