VoronDesign/Voron-2 #150: Unable to print TPU

bjoern79de, 2020-06-21:
I had no issues with Afterburner-beta, but using Voron2.4 I'm unable to print TPU.
The Filament-Path around the drive gear seems to be wider than in the first version, so the filament is pressed into the gap.

Maglin, 2020-06-21:
which extruder gears are you using?  Some have a smaller small radius than others and this could explain your problems.  

FYI I've printed TPU with no problems but it also was Shore 98 stuff.

bjoern79de, 2020-06-21:
I'm using this one:
https://www.aliexpress.com/item/32917029058.html?spm=a2g0o.productlist.0.0.772b5a5eL9DykH&algo_pvid=a25b5428-e61c-4c96-95c4-7ad0b6da9c6b&algo_expid=a25b5428-e61c-4c96-95c4-7ad0b6da9c6b-0&btsid=0b0a0ac215927619359532812e6485&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

insurgus, 2020-09-25:
I recommend checking on Discord and seeing if other users have had this issue. There doesn't seem to be much traction here unfortunately.

closed 2020-09-25 by insurgus

laserbeak, 2021-09-25:
Did you ever get an answer?

newphreak, 2021-09-25:
There is this if you want to venture into tpu:
https://thangs.com/francismarasigan/Voron-AfterBurner-ClockWork-Feeder-Modstl-24906