VoronDesign/Voron-2 #155: SSR wiring in 2.4 Assy Manual incorrect

AeroAggie, 2020-06-26:
Page 117 at the SKR end of the diagram, positive is shown on the right and negative on the left.  Should be reversed per SKR pin map.  

The diagram in Eddie's SKR setup guide is correct.  My bed failed to heat on config check when wired per the manual, but heated correctly when wired per Eddie's diagram.

david0618, 2020-08-30:
AeroAggie is correct.   I have the manual but I also have the github SKR 1.3 setup and they are both different in the instructions.    At first my heater would not work and after looking at my wiring and the directions I saw this.  So did a little more research and found this post.    Eddies setup is the correct one

insurgus, 2020-09-25:
If you're talking about the SSR Low Voltage diagram, this has been updated.

closed 2020-09-25 by insurgus