VoronDesign/Voron-2 #161: V2.4 Manual Correction

origuy42, 2020-07-04:
Page 27 appears to have two minor errors that may cause new builders slight confusion (like me!). They are easily overcome, and maybe it's just me, but clear/correct instructions >mostly correct instructions:

1) The M5x30 SCHS in the 'eyeholes' are mislabeled (at least according to the CAD file). Should be labeled BCHS instead. Not sure if there are clearance issues here, but consistency between the manual, BOM and CAD is probably a good thing.

2) The order of assembly on the page is physically impossible. The M3x16 SCHS can't go in before the lower printed part.

Luddi1, 2020-07-05:
Another minor thing I spotted: On Page 59 you need a M3x30 instead of M3x20.

kauedg, 2020-07-16:
In the Introduction's "Final Thoughts" the text is incomplete.

[image]

kauedg, 2020-07-22:
[image]

insurgus, 2020-09-25:
> 
> 
> Another minor thing I spotted: On Page 59 you need a M3x30 instead of M3x20.

I think this was already resolved. The page numbers are all whacky now due to addition of sections. The carriage fasteners were updated to M3x30 some time ago though.

insurgus, 2020-09-25:
> 
> 
> In the Introduction's "Final Thoughts" the text is incomplete.
> 
> [image]

This has been fixed.

insurgus, 2020-09-25:
>     1. The M5x30 SCHS in the 'eyeholes' are mislabeled (at least according to the CAD file). Should be labeled BCHS instead. Not sure if there are clearance issues here, but consistency between the manual, BOM and CAD is probably a good thing.

Fixed.

insurgus, 2020-09-25:
>     2. The order of assembly on the page is physically impossible. The M3x16 SCHS can't go in before the lower printed part.

Need clarification on this.

closed 2020-10-28 by insurgus