VoronDesign/Voron-2 #162: Wrong M2 self tapping screw in BOM

lazymaniac, 2020-07-05:
Hey

In BOM I found M2x10 listed, but in instruction manual says about M2x8. M2x10 are too long for DIN mounts because they go all the way through.

lazymaniac, 2020-07-05:
Also the amount is incorrect. Instruction requires 16 screws (2 - z endstop, 4 - xy endstops, 2x2 - skr mounts, 6 - rpi) but in BOM - 10

avongil, 2022-01-10:
The mcmaster screws are too short to attach the microswitch reliably. The head is also way too big for that application if you want to lay them flat and solder the JST direct. I had to source other ones that had a thinner head and longer body for all the end stops.   The din rails where ground flush.

avongil, 2022-01-10:
The Digikey switches have holes that are a tiny bit larger than 2mm, so it would not hurt to keep the McMaster linked screw at 1.91mm. The linked McMaster screw is an M1.91 https://www.mcmaster.com/99397A267/   - these screws I was not a fan of. Note how big the head is - 4mm. 
 
 I propose either one of these - they are a much better fit for the switches and will work everywhere:   
 
 https://www.mcmaster.com/99461A620/
 https://www.mcmaster.com/96001A127/
 

Current screws in the BOM from McMaster (head diameter too large):
[image]

[image]

DigiKey microswitches on the BOM have holes that are 0.081" in diameter - 2.06mm.  

[image]

avongil, 2022-01-15:
Confirmed the mcmaster part: https://www.mcmaster.com/99461A620/
Is a way better fit, and also easier to cut if you have to although it looks like the new style din connectors are made for it. Great job Steve!  Gorgeous and well labeled parts.  
[image]

I did not want to bust off the flap, but this fits way nicer than the official linked mcmaster part in the microswitch.  3.5 fully engaged threads. 
[image]
[image]

closed 2022-02-23 by raymondh2