VoronDesign/Voron-2 #179: Incorrect Bowden fitting in sourcing guide

shiftingtech, 2020-08-07:
The sourcing guide link (V2.4 tab, line 73) for a bowden fitting goes to the embedded bowden fitting as used in Jetpack, rather than the threaded one used in the V2.4 Exhaust assembly.

translucentfocus, 2020-09-15:
I just looked into this and can confirm @shiftingtech discovery.

**Recommended Addition to the Voron 2.4 Sourcing Guide Tab**

| Category | Component                | Qty | Recommended                        | Alternative Source                                       |
|----------|--------------------------|-----|------------------------------------|----------------------------------------------------------|
| Bowden   | Threaded Bowden Coupling | 1   | [McMaster-Carr](https://www.mcmaster.com/5449K171/) |  [E3D](https://e3d-online.com/products/threaded-bowden-coupling) |

Btw, I think this is a brilliant ~use~ **recycling** ♻ of parts from back in the 2.1 days 👏

I checked the following sourcing guides and the part is not listed on any of them:
❌ Voron 2.4 Sourcing Guide
❌ Afterburner Sourcing Guide
❌ Mobius 3.1 Sourcing Guide
❌ JetPack Sourcing Guide

**Image taken from Voron 2.4 Manual**
[image]

FHeilmann, 2020-12-14:
Thanks guys, should be fixed now. If not, feel free to reopen this!

closed 2020-12-14 by FHeilmann