VoronDesign/Voron-2 #182: I want to provide some help for Voron

FLYmaker, 2020-08-16:
I want to provide some help for Voron. I see that most of you use multiple motherboards. I want to provide you with some FLY motherboards for free, supporting 9 drivers. Please give me a contact

vallentinc, 2020-08-16:
> I want to provide some help for Voron. I see that most of you use multiple motherboards. I want to provide you with some FLY motherboards for free, supporting 9 drivers. Please give me a contact

Thanks for the offer, but we cannot provide you our contact here because of spam. 
Are you a member of our Discord channel? If yes, you should make the announcement there and/or request a vendor thread like Powge did.

drasrx, 2020-08-16:
I'm not sure who @vallentinc is, but, he's not able to comment on behalf of the dev team as far as I am aware.

@FLYmaker please send an email to operations@vorondesign.com, and one of the dev team will discuss with you.

Thanks!

FLYmaker, 2020-08-16:
> 我不确定@vallentinc是谁，但是据我所知，他无法代表开发团队发表评论。
> 
> @FLYmaker，请发送电子邮件至[operations@vorondesign.com](mailto:operations@vorondesign.com)，开发团队之一将与您讨论。
> 
> 谢谢！
Thank you

FHeilmann, 2020-11-17:
Closing since this query seems to be resolved. Feel free to reopen if you have further questions.

closed 2020-11-17 by FHeilmann