VoronDesign/Voron-2 #186: 2.4 BOM is short 3x M5x16 bolts

jrosen9, 2020-09-15:
I ran the configurator for a direct feed, 300mm with Misumi corner cubes.  The produced BOM stated I needed 11 M5x16 screws.  Following the manual, I have needed 14 of these of these screws

julianschill, 2020-11-29:
It is actually short on 4x M5x16. In the manual there are 2 on page 51, 4 on page 52, 4 on 54, and 1 for the spool holder. Then you need 4 for the bed and 16 for the frame for blind joints. That makes 31 for blind joints and 15 for corner cubes.

julianschill, 2021-05-28:
The BOM is indeed short of 4 M5x16. Here is an image of where the M5x16 are used: 
[image]

cyrix2000, 2021-11-19:
Why is the BOM now showing just 11 M5x16 BHCS bolts?

closed 2022-02-23 by raymondh2