VoronDesign/Voron-2 #188: Different IGUS chains in BOM and Sourcing Guide

kage-chan, 2020-09-15:
Hi,

if IGUS chains are used in the configurator, BOM calls for IGUS e2 chains for x and y, which open on the outside. Sourcing Guide on the other hand calls for e2i, which opens on the inside. I personally prefer the inside opening variant for x and y, but whichever is preferred by the dev team, BOM and sourcing guide should match up.

thejobbitt, 2020-12-11:
The lengths are also slightly off, causing confusion. The links are 20mm so the lengths should reflect that. 

250mm
z - 22 links, 440mm [1.4483 ft]
xy - 16 links, 320mm [1.0533 ft]
300mm
z - 25 links, 500mm [1.6458 ft]
xy - 19 links, 380mm [1.2508 ft]
350mm
z - 28 links, 560mm [1.8433 ft]
xy - 21 links, 420mm [1.3825 ft]

*Note the feet lengths are not a direct conversion, but pulled from the Igus site as their "chain length" output when inserting the travel length.

raymondh2, 2022-02-23:
Addressed with V2.4r2

closed 2022-02-23 by raymondh2