VoronDesign/Voron-2 #190: [V2.4] Manual and Voron2_SKR_14_Wiring.pdf are inconsistent for A/B motor

elpopo-eng, 2020-10-07:
Hi,

The manual and Voron2_SKR_14_Config.cfg indicate connection of motor B on X and motor A on Y.
[image]
[image]

But in Voron2_SKR_14_Wiring.pdf the motors are inverted
[image]

Thanks

mateuszbrzezinski, 2020-10-07:
I had same issue.
The other issue is that the extruder by default has wrong direction, at least 2 persons (including me) had to update the configuration to make it work.

insurgus, 2020-10-28:
Updated. Will PR shortly.

closed 2020-10-28 by insurgus