VoronDesign/Voron-2 #212: Need more sources and specs for linear rails

ax00b3, 2021-02-11:
The configurator only says "Linear Rail MGN9H" and the sourcing guide only lists RobotDigg, Zyltech and CNA. Can we have the THK and/or genuine Hiwin part numbers?

jdlongenecker, 2021-02-11:
Hello @ax00b3! We don't recommend purchasing THK and/or Hiwin rails as the added benefit you get is minimal vs. the significant cost increase. If you wish to use those rails on your printer, feel free to look up the part numbers/drawings on their websites and check if they offer anything compatible with the MGN9H rails commonly found elsewhere. 

Thanks!

ax00b3, 2021-02-12:
Well in my case it's because I'm working with a client that strongly prefers USA/Japan parts and wants to avoid any Chinese/no name brands.

raymondh2, 2021-02-12:
Those of us on the dev team are hesitant to recommend or add anything to the sourcing guide we have not personally tested. As few, if any, of us have tested other rails we will not be adding them to the guides.

closed 2021-02-12 by raymondh2