VoronDesign/Voron-2 #213: Door panels have no tolerance

julianschill, 2021-02-15:
When ordering everything to spec, the door have zero tolerance and don't close properly, especially when getting warm. They block each other in the middle and at the latches. 

I would suggest to specify the doors 0.5mm smaller in both directions.

Alternatively the hinges and latches should account for that.

NameOfTheDragon, 2021-05-16:
Let's not be hasty, I like the snug fit. I didn't like it to begin with but it has grown on me. On my build the panels line up perfectly in the middle and its almost as if you have a single piece across the front. I find this desirable from the point of view of draft exclusion and keeping in the heat. It does take a little jiggle to get them to close like that but now that I'm used to it I'm OK with it.

[image]

There's always the option to trim them down a bit or sand with a sanding block, it's better to start with a tight fit and sand down than have too big a gap, although I suppose you do have some leeway at the hinges.

claytono, 2021-05-18:
I've had this problem also.  Both sets of door panels I've ordered from PrintedSolid have met spec, but had slight interference issues.  If they were 0.5mm smaller, then you could mount them on the hinges closer to the center if you prefer that, but with the current size, that's not an option.

stvptrsn, 2021-05-19:
Most cut to order shops specify a cut tolerance.  For instance this is the text from Tap Plastics "Cut to size orders are cut with table and CNC panel saws and have a cut tolerance of +/- .060”. Plastic sheets have a thickness tolerance of +/- 10% and can have thickness variations throughout the sheet. Cut to size items cannot be returned for a refund or exchange."  Accounting for this in CAD is not reasonable and you should expect to check your panels before install and adjust as necessary.

julianschill, 2021-05-19:
A solution could also be to design adjustable hinges, so you can move the doors around a few 0.1mm until it fits and compensate for the tolerances of the cutting.

claytono, 2021-05-19:
@stvptrsn I assumed this was more of a BOM than a CAD issue.  Are the BOM dimensions adjusted for that sort of thing?

stvptrsn, 2021-05-19:
> @stvptrsn I assumed this was more of a BOM than a CAD issue. Are the BOM dimensions adjusted for that sort of thing?

I don't think so.  But it's hard to compensate for what could be a 3mm swing based on advertised cut tolerances.

claytono, 2021-05-19:
Maybe it would be helpful to call this out as a note in the sourcing guide.  I had a similar problem with my V0 panels I ordered from TAP and it never occurred to me that it might be an issue.

insurgus, 2021-06-01:
Putting a note in the 2.4 manual regarding fitment.

closed 2021-06-10 by raymondh2