VoronDesign/Voron-2 #214: RE:  M4 Knurled nut for bed

bigjoe4265, 2021-02-15:
In the Assembly .pdf and BOM for the 2.4 you have a M3x16 SHCS (.5mm thread pitch) screwing into a M4 knurled nut (.7mm thread pitch) to anchor the bed.  Shouldn't this be an M3 knurled nut (thread pitch is .5mm)?

raymondh2, 2021-02-15:
No this needs to be m4 as we aren't using fit as a nut, just a space.

closed 2021-02-15 by raymondh2

bigjoe4265, 2021-02-15:
> 
> 
> No this needs to be m4 as we aren't using fit as a nut, just a space.

Do you mean for thermal expansion?  or height of the thumb screw.  It's roughly .008 of an inch height difference between the M4 and M3.  The concern of course is stripping the bolt due to the difference in thread pitch.

raymondh2, 2021-02-15:
It's being used as a spacer. Not as a nut.