VoronDesign/Voron-2 #235: Bowden retainer does not fit over a T-nut

NameOfTheDragon, 2021-05-10:
I note that the door hinges were updated a few days ago so that they now fit over a "roll-in" T-nut. The Bowden retaining clip suffers from the same issue.

NameOfTheDragon, 2021-05-10:
Never mind - it was probably intended to use hammerhead T-nuts with the hinges and clips.

closed 2021-05-10 by NameOfTheDragon

jurisv, 2021-05-10:
Please note that original Misumi nuts with clip on the side are the same length as the hammer-head T nuts. Unless cheap Chinese nuts with ball are used, there should be no issue. In fact those roll in ball spring nuts are so bad quality that they should no longer be on the list. I purchased ones that were in the sourcing guide last year, and again this spring. They all have manufacturing defect where the width of the nut is too wide for proper insertion. 8mm+-0.1mm vs 8.3..8.4mm actual size. I reordered from multiple vendors on Amazon and they all were the same wrong size.

aureumlocus, 2021-05-10:
I got a Formbot kit a month ago. I'm aware of the limitations of the kit but hadn't seen the T-nut issue before. I just checked and the T-nuts I have (M3 and M5) at 8.13 and 8.11 mm wide respectively are just outside the tolerances you mentioned. I'm hoping they'll be OK. I've just started assembling the frame and have just inserted the M5 nuts for the heated bed. They went in relatively easily. I haven't tried to get them out though! [image]