VoronDesign/Voron-2 #237: [V2.4] SKR 1.4 AB motor pinout different in config vs guide

cbc02009, 2021-05-15:
The motor connections in the guide for AB motors are opposite what they are in the config:

wiring guide (pg 2):
[image]

config:
```
[stepper_x]
##	Connected to X on mcu_xye (B Motor)
...
[stepper_y]
##	Connected to Y on mcu_xye (A Motor)
...
```

Someone pointed this out on discord so I figured I'd create an issue.

insurgus, 2021-06-01:
Some discussion has to be had internally to decide how we are going to standardize this. We used to use A/B and then added X/Y and it has caused some confusion in the community. That coupled with the fact that the configs call for X/Y/Z for the MCU is certainly something we should discuss.

stephenlawrence, 2021-07-14:
I prefer xyz vs. A/B myself.

NameOfTheDragon, 2021-07-14:
> I prefer xyz vs. A/B myself.

Z makes perfect sense but X/Y does not make sense in the context of a CoreXY mechanism, since both motors cause movement in both X and Y. Only by moving both motors is it possible to move in just X or just Y. So which one do you call the X motor and which one the Y motor? It has to be "neither". A/B doesn't really make much sense either because that still doesn't make it obvious which motor is being referred to unless you look it up on a diagram. In my view, it makes more sense to call them "Left" and "right", because one of them is on the left and the other is on the right. That's still a bit ambiguous because it depends which direction you are looking from, so I would say it is even more clear to call them "Back left" and "Back right" or "BL" and "BR". I think thats as good as it gets in terms of disambiguation and having a name that is "intuitively obvious".

stephenlawrence, 2021-07-14:
Ok, I changed my preference now to BL and BR. :)

cbc02009, 2021-07-15:
I agree back left and back right (BL and BR) makes the most sense.

mathiasm74, 2022-02-17:
I've also noticed this issue. I connected the A motor (BR) to MOTOR0 and B motor (BL) to the MOTOR1 on my Octopus, and the toolhead homed away from the x endstop. The [only solution](https://docs.vorondesign.com/build/startup/#v2-1) to this is to swap the wires. This seems wrong, going by what the documentation says. 

But I'm confused about the source of this problem. I would assume the Voron documentation is correct, but this yields the wrong result. I've noticed that the motors in the illustration appears to point in the wrong direction compared to how they are on a 2.4. Is it possible that they were reoriented compared to an earlier version, and that the A and B has effectively swapped places?
 
[image]

closed 2022-02-23 by FHeilmann