VoronDesign/Voron-2 #242: Probe malfunctioning crash prevention routine

knackmanic, 2021-06-12:
A simple routine could prevent a crash due to a dead probe.

After implementing this feature, I believe this is something that might be useful for others considering the doubtful QC and disparity of the probe that are used on the Voron 2.4.

a1co, 2021-06-13:
That sounds like a good idea. I cannot check now, but i belive my probe sould light ub when the z endstop is triggert. Then that would not cost any additional time.

julianschill, 2021-06-13:
This can be done with a macro. Check my CHECK_FLEXPLATE macro here: https://github.com/julianschill/CeBe_config/blob/cafaae847d028ac051ec81ee0b90436a6cc46ce3/macros.cfg#L66

a1co, 2021-06-13:
> 
> 
> That sounds like a good idea. I cannot check now, but i belive my probe sould light ub when the z endstop is triggert. Then that would not cost any additional time.

That was nonsense: The z endstop is behind the bed. But there could still be a screwhead for probe testing placed.

knackmanic, 2021-06-13:
> 
> 
> This can be done with a macro. Check my CHECK_FLEXPLATE macro here: https://github.com/julianschill/CeBe_config/blob/cafaae847d028ac051ec81ee0b90436a6cc46ce3/macros.cfg#L66

Yup, that's exactly what I meant by routine (including a macro that does the check in the config). IMO, it'd be a good thing to add in the doc (maybe like a "safety" configuration section?) I've seen a few user in the discord losing their probe and digging a nice hole in the plate. The homing sequence could also be done at lower Z current which would help prevent catastrophic issue  in case your forget to put the Z pin, the head got stuck somewhere during the homing and it's not underneath the Z pin / other bad luck, etc.

raymondh2, 2021-09-20:
Inactivity

closed 2021-09-20 by raymondh2