VoronDesign/Voron-2 #255: AB Idler Belt Wear

chrisingham05, 2021-08-14:
I have a suggestion for improving the AB front idlers as I've had some damage to my belts due to how they are designed. Currently, the idlers are tightened using a screw at the front which are aligned to the centre of the idler, however as the belt idler is either at the top or bottom depending on which side it sits this creates a rotational force which mis-aligns the idler bearing giving it a slight angle. This angle causes the belt to rub at one end causing wear to the belt. I suggest moving the screw so that it is in line with the centre of the idler bearing so that there is no rotational force. For my printer, I plan to make a few spacer wedges that I can stick in the idler to stop this rotation, but just moving the tensioner screw in the design would fix this quite easily. Only downside is it would not look symmetrical from the front.

[image]

chrisingham05, 2021-08-14:
Slight damage to the belt found so far

[image]

chrisingham05, 2021-08-17:
Temporary solution is to measure the gap between the red carriage and back of the idler with callipers to see the distance when the belts are tensioned, then printed small spacer blocks for each side to align the red carriage. Using one block at the end of where the belt meets allowed me to change the angle of the red carriage by tightening or loosening the screw. Appears to have improve the situation for now. Obviously on the right the block is at the top and on the left the block is at the bottom. Note, the picture is the correct way up, I assembled the red carriage upside down but that doesn't affect the printer or this issue. 

[image]

ServAlex, 2021-08-17:
I also experience this issue on B belt, haven't verified culprit yet

yonkiman, 2021-09-04:
I have the exact same wear issues around front idlers on A and B belts.  Your explanation sounds right - may try to address by twisting idler frame forward or backward before clamping down with vertical pin.  Would like to address before I re-belt…

chrisingham05, 2021-09-06:
Since fitting the spacer blocks I've not noticed any more wear on the belts

1ucket, 2021-09-06:
Have the same issue, for me it even cut a notch into the belt through which it is now leaving the idler.
[image]

DanJunior78, 2021-09-06:
Would be a good idea to redesign it, as you mentioned, test it and do a push request.
I'll soon build my second V2 and i could also try it.
In my first I've same issues.

ServAlex, 2021-09-06:
> Have the same issue, for me it even cut a notch into the belt through which it is now leaving the idler.
> [image]

this one is really strange, I don't know what part could do that, the only guess is the flange of the idler, but to do that your belt have to be on top of it which would be quite noticeable.

1ucket, 2021-09-06:
> 
> 
> > Have the same issue, for me it even cut a notch into the belt through which it is now leaving the idler.
> > [image]
> 
> this one is really strange, I don't know what part could do that, the only guess is the flange of the bearing, but to do that your belt have to be on top of it which would be quite noticeable
> 
> Ohh, I see a line in the middle of my belt too, guess it's where 2 bearings meet in the stack

It would become nearly completely stuck in the rear left corner (aka the part where the belt would leave the idler). So yeah, it was quite noticeable.

GadgetAngel, 2021-09-26:
> Temporary solution is to measure the gap between the red carriage and back of the idler with callipers to see the distance when the belts are tensioned, then printed small spacer blocks for each [...]

Where is the parametric .stl file for the red carriage spacer blocks that I can print out for when I do my Voron 2.4 build?  I really do not want to damage by High Temperature Gates Belts!

1ucket, 2021-10-21:
> 
> 
> I redesigned the A and B idlers to accommodate this change, will print and test.
> 
> [image]
> 
> Not sure how or where to share the STLs though!

There's a [guide ](https://docs.vorondesign.com/community/support/pull_request_guide.html) on how to add your mods. As @DanJunior78 said just add it to VoronUsers/tree/master/printer_mods.

netweaver1970, 2021-10-21:
Another (more adjustable and skinnier solution) https://github.com/Ramalama2/Voron-2-Mods/tree/main/Front_Idlers

GadgetAngel, 2021-11-27:
> Another (more adjustable and skinnier solution) https://github.com/Ramalama2/Voron-2-Mods/tree/main/Front_Idlers

yeah, I just found this suggestion.  Why doesn't Ramalama2 submit it to the VoronUsers mods branch?

I also see that Ramalama2 has the Front_Idlers offset for the GE5C mod and his design also uses a PIN in stead of a M5 bolt to align the idlers.  Thanks for the suggestion.

DaddyBender, 2021-12-04:
Please excuse me if I don't have the terminology quite right.

I was going over the latest f3d, and I think I found belt-to-belt contact?  
Example:
If I remove the exterior parts to look at the idler bearings, it looks like the z belts are rubbing against the A/B (?) belts?

[image]

GadgetAngel, 2021-12-04:
> Please excuse me if I don't have the terminology quite right.
> 
> I was going over the latest f3d, and I think I found belt-to-belt contact? Example: If I remove the exterior parts to look at the idler bearings, it looks like the z belts are rubbing against the A/B (?) belts?
> 
> [image]

I was looking at the belts of the Voron 2.4 assembly model also.  If you look the belt path does not fit the idlers, I think the model has an error in it.  To redraw the belt path to make if fit as it should would take time. 

But I agree the model does show the belt path is off in places.

Galifrey1965, 2021-12-22:
I reprinted my tensioners today so they fit tighter around the carrier and after nipping up the M5 bolt they sit square no problem. However, seeing as this is down to print tolerances and poor design, I can see how Chrisingham05's design is much better and avoids running the bolt through the split in the joined carrier. Overall I cannot help but wonder how the design team missed this as a potential weakness.

closed 2022-02-23 by FHeilmann