VoronDesign/Voron-2 #262: About a / b motor and X / Y axis transmission principle

jack9603301, 2021-09-13:
Hi, I am deeply analyzing and trying to improve and DIY a Voron 3D printer at a lower cost, but we don't seem to understand the principle of A/B motor and X-axis/Y-axis transmission. From the PDF, A/B motorcan control the extruder at the same time, but we can't see how to control the movement of X-axis/Y-axis

AaronHerrick, 2021-09-13:
https://www.youtube.com/watch?v=8WLZ8OesMF4

DunarDolorgiet, 2021-09-13:
The Voron2 belt path is an implementation of the "CoreXY" pattern. You can read about the background over here: http://corexy.org/theory.html

closed 2021-09-13 by DunarDolorgiet

jack9603301, 2021-09-13:
@DunarDolorgiet So what should I consider when writing klipper configuration files? I might consider changing the main control board