VoronDesign/Voron-2 #263: VORON2_v2.4_Assembly_Step

WTFacker, 2021-09-14:
Please can you redownload the VORON2_v2.4_Assembly_Step archive. Thank you!

MtPegasus, 2021-09-24:
When extracting the Step file I get an error.

javierdc, 2021-10-01:
I'm also having issues with the zip file. I have removed and recloned the voron 2.4 branch but same issue.

MrDix, 2021-10-03:
Same situation on my side . Zip file is reported to be broken by 7zip and WinRAR. If your ignore the error and open the STEP assembly anyways there are a few sub-components missing in the assembly. For generating the STEP assembly from the fusion360 sources a commercial Fusion360 license is required. The personal/hobby use free license is lacking the STEP export option.
A reupload or reexport+reupload of the STEP assembly would be highly appreciated. 
Thank you for your awesome work!

MrDix, 2021-10-03:
OK, I see there is a new version of the zip file: https://github.com/VoronDesign/Voron-2/blob/f8f5a8f4f0ad763af43dd2ddcf5b74afd33a6182/CAD/VORON2_v2.4_Assembly.step.zip and this can be unpacked without problems. This new and fixed version is just not available in the Voron2.4 branch jet and needs to be fetched manually.

FHeilmann, 2021-10-08:
I merged the updated version this morning. Sorry for the delay!

-Florian

ghent360, 2021-11-27:
Latest step file is missing one of the X carriage link components.
[Image from Onshape STEP import]([https://photos.app.goo.gl/aJfPb8EMWicjFL6L6)

ghent360, 2021-11-28:
Please disregard my last comment. This appears to be a bug in the onshape importer. I exported the missing part and it says "failed to process" on the file.

bfschris, 2021-12-04:
Hi all - I'm still getting the STEP file import error:

Error: Encountered invalid record index.
STEP-in did not complete successfully.

even when I try the link above from Oct 3.  7-zip reports no errors with zip file and it seems to extract fine; I just can't open the STEP.

anyone else having trouble?  thanks in advance -

bfschris, 2021-12-04:
to clarify my comment, above: I tested ZipFix#265 with no success.

raymondh2, 2021-12-18:
I've tested the current zip file having unzipped it on two different operating systems (Ubuntu 20.4, Windows 11) with their native decompression tools. I have also imported it into three different CAD packages successfully (Fusion 360, SolidEdge 2022, and SolidWorks 2020.) Until I can recreate the issue I am closing this.

closed 2021-12-18 by raymondh2

bfschris, 2021-12-19:
Hi Raymond- thanks for looking into the problem and I apologize for the trouble.  It must be me.  I will use one the CAD options you listed and report back, to help close the loop for others.  Best regards- Chris