VoronDesign/Voron-2 #264: About STL File and BOM list selection

jack9603301, 2021-09-14:
Hi, I am studying Voron 2.4 BOM list and STLs, I am trying to build a 350x350 3D printer, I found that the bom generated has a 9mm GT2 pulley, but I may only be able to buy a 10mm pulley, do I have to find a 9mm pulley? To Install into the STL z drive.

NameOfTheDragon, 2021-09-14:
This isn't the right place to ask general questions. There is a Discord server and a Facebook group, both of which would be more appropriate.

I expect the 9mm refers to the width of the belt. The Z drive uses 9mm belts and the A/B drives use 6mm belts. It should be easy to get those pulleys, they are very common.

raymondh2, 2021-09-14:
For whatever reason they sell pulleys for 9mm belt as 10mm pulleys. So getting a 10mm pulley for 9mm belt is correct

raymondh2, 2021-09-20:
@jack9603301 does this answer your question?

closed 2022-01-11 by raymondh2