VoronDesign/Voron-2 #285: Fragile shaft

jack9603301, 2021-12-18:
Why can't we improve your shaft to increase its strength or reliability? I don't know if it often breaks due to printing problems. Can't we design our shafts as separate? Alternatively, a metal shaft is recommended.

During maintenance, it is often possible to open the door 180 degrees, which is easy to break in such a large angle and frequent operation.

raymondh2, 2021-12-18:
Which shaft? There are many shafts used in the build. If you are refering to the Afterburner hinge it is adequet as is and we have not heard reports of it breaking when ABS is used as the pin

jack9603301, 2021-12-19:
I mean the front door

> Which shaft? There are many shafts used in the build. If you are refering to the Afterburner hinge it is adequet as is and we have not heard reports of it breaking when ABS is used as the pin

raymondh2, 2021-12-19:
This has not been an issue reported by anyone. I am closing this as a non issue

closed 2021-12-19 by raymondh2

julianschill, 2021-12-19:
There is also a Mod for hinges with M3 screwsas shaft.
https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/mosher/m3_door_hinges

jack9603301, 2021-12-19:
> There is also a Mod for hinges with M3 screwsas shaft. https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/mosher/m3_door_hinges

https://github.com/VoronDesign/Voron-2/issues/285#issuecomment-997437067

I'm watching.

https://voronregistry.com/mods/270_clampinghinges_alexandert-moss