VoronDesign/Voron-2 #291: Problems with the provided .stl files

Schmiddy84, 2022-01-16:
Hello everyone, I have problems with some .stl files. Everything is displayed correctly in the slicer, but in some layers I have errors when printing, such as printing straight through a hole or a radius is missing in the outer contours of a layer.

I can avoid these errors by creating the .stl files myself from the supplied .step file. However, I now have problems with generating the "skirt" parts myself because the .step file does not seem to show a size of 350. How do I get the CAD files from the 350mm version. Many Thanks.

Greetings Schmiddy84

GadgetAngel, 2022-01-16:
> Hello everyone, I have problems with some .stl files. Everything is displayed correctly in the slicer, but in some layers I have errors when printing, such as printing straight through a hole or a [...]

@Schmiddy84 The Skirt files for the 350mm build are in the Fusion 360 (or .f3d file) for 250mm and in the STEP file for the Voron 2.4.  If you are using the .f3d files for the Voron 2.4 , then you will have to go to the "lower valance" and select the skirt of interest.  The 250mm will be displayed, but if you goto the the "Body" folder of the skirt you will see three bodies included in the item. In the Skirt Body, only one body is visible (the 250mm is visible). Just turn off the 250mm skirt and turn on the 350mm skirt.

If you are using the .step file for the Voron 2.4, then all the skirt bodies should be turned on, because the .step file will not save them if they are not visible.  So If you can not follow what I said to do above, just down load the Voron 2.4 model in .STEP format and you will see all three skirts turned on all the way around the printer.  

I hope this helped.

Schmiddy84, 2022-01-16:
Thank you for your help. I had the step file open the whole time and couldn't find the 350mm option. Now that I opened the .f3d file I could find the hidden skirts.

Schmiddy84, 2022-01-16:
> @Schmiddy84 The Skirt files for the 350mm build are in the Fusion 360 (or .f3d file) for 250mm and in the STEP file for the Voron 2.4. If you are using the .f3d files for the Voron 2.4 , then you [...]

I was able to find all skirts except for the front/rear skirt b. Strangely enough, this can only be found in the 250mm version.

Schmiddy84, 2022-01-16:
[image]

closed 2022-01-16 by Schmiddy84