VoronDesign/Voron-2 #292: V2.4 BTT Octopus printer.cfg - [stepper_z] motor direction should be reversed

310weber, 2022-01-18:
@cbc02009 
Z stepper motors should alternate directions.  In the current [V2 Octopus config example](https://github.com/VoronDesign/Voron-2/blob/Voron2.4/firmware/klipper_configurations/Octopus/Voron2_Octopus_Config.cfg) both Z and Z1 rotate in the same direction.  This causes Z to move down when Z1-Z3 move up (and loud belt slipping noises).  

[stepper_z] should be reversed.

Current:
```
[stepper_z]
step_pin: PF11
dir_pin: PG3
enable_pin: !PG5

```

Should be:
```
[stepper_z]
step_pin: PF11
dir_pin: !PG3
enable_pin: !PG5
```

cbc02009, 2022-01-18:
The inversion/non-inversion of any of the steppers in the default config is purely for example purposes. It can change from stepper to stepper as the are wired differently between vendors and even sometimes with motors from the same vendor. This is makes it impossible to have one stepper config that works for everyone.

The startup documentation details the process for adjusting the z steppers so they all move in the same direction.

310weber, 2022-01-19:
I get that, but the most common scenario is that a builder will use one brand/model of stepper and wire them all the same way. In this scenario the Z motor directions should alternate (e.g. Z & Z2 reversed and Z1 & Z3 not reversed). That way at least they will all move up or all move down together. I think it makes sense to have the example config align with the most common scenario rather than the unlikely scenario that someone happens to wire the Z motor backwards compared to all the others. 

In my case I just happened to miss that one motor moving down rather than up first during the stepper_buzz step. So, when I went to do X homing one corner of the gantry moved down while the other three moved up. It made horrible belt skipping noises. Scared the crap out of me. Totally my fault for missing it, but I'd bet this simple change would save others from the same fate.

cbc02009, 2022-01-19:
that's a fair point. I'll work on PR.

PR #293 (2022-01-19): fix default z stepper directions for octopus config.
fixes #292
closed 2022-01-19 by KiloQubit