VoronDesign/Voron-2 #294: Uneven / Incosistant layers on X-axis

kubusAB33, 2022-01-21:
## Problem:

Uneven / Incosistant layers on X-axis, while the proplem does not occur on Y-axis.

Printer: Voron V 2.4 
Hardware all stock
Board: BTT Octopus
Stepper Drivers: TMC 2209  

## Trials to solve:

-	Checked all idlers/pulleys if they can rotate freely and without play. Check
-	Checked all screw if sitting tight (gantry, hotend, frame). Check
-	Did Z-joint procedure (losen M5 bolts, do QGL, fasten screws afterwards) sveral times. Check
-	Tried differend belt tensions. Check
-	Check if belts are eventy tensiond. Check.
-	Printed test part on several bed locations. Check
-	Tried printing with differend speeds and accelerations. Check.
-	Tried printing with and without Input Shaper. Check
-	Tried printing with and without Preassure Advance.
-	Doublechecked toolhead if there is any wobble/play. Check
-	Cleaned nozzle. Check
-	Checked print bed for play/wobble. Check
-	Checked if toolhead is moving in X when gantry is moved in Y only (maually). Check.
-	Checked x-rails for propper alignment. Check.
-	Checked different filaments and print temperatures. Check.

Now I ran out of ideas. 
Since I checked all the mechanical parts, I doubt, that the problem is caused by them, even if it clearly looks like a mechanical issue.

### Left over ideas:

-	Something is wrong with the synchronization of A and B motor
-	Something is wrong in the firmware I use
-	Something is wrong with the connections of the motors
-	Something is wrong with the current settings oft he steppers
-	There may be minimal play in Y, when toolhead moves in X direction

That’s just my thoughts. I hope for some input, support and help for solving my problem.

Thanks a lot in advance!

[image]
[image]
[image]

[image]
[image]

kubusAB33, 2022-01-21:
Here's an example of a cylinder.
[image]

closed 2022-01-22 by kubusAB33

aaron-hentschel, 2023-06-11:
I have this exact same issue on my Voron 0.2 (although not as bad as kubusAB33).  What was the fix for this?