VoronDesign/Voron-2 #305: Octopus board and hall effect XY end stop board

bubnikv, 2022-02-08:
Please document that:

1) The Octopus board end stop inputs are 5 volts tolerant, thus it is possible to connect a 5V end stop input in there. I suppose other STM based boards are similar.

2) The hall effect XY sensor requires 5V, however it outputs 3.3V. Formbot provided cable powers the board from 3.3V. Please document that the builder may need to split the cable provided by some kit providers to power from 5V, not 3.3V

bdbell, 2022-02-08:
This would fit better in the documentation site.

bubnikv, 2022-02-09:
Regarding wiring: There is a risk that any of the mains wring including the bed heater wire breaks off and touches the metal frame.  The metal frame shall be grounded.

raymondh2, 2022-02-09:
As documented you should follow your local code when it comes to mains wiring. As we can't document it all we don't. We put in general best practices. The first two issues are more docs repo issues. 

Hope that explains it for you :)

closed 2022-02-09 by raymondh2

bubnikv, 2022-02-09:
> As documented you should follow your local code when it comes to mains wiring. As we can't document it all we don't.

This is one that I strongly disagree with. Grounding a metal frame is something one shall do. I wonder how many Voron builders did not do that.

According to a local code, most likely the vast majority of the Voron builders are prohibited to do the mains wiring themselves. We cannot expect the builders to study the code. General safety guidelines should be given.

bubnikv, 2022-02-10:
The SSR shall be grounded through its DIN rail. How many of Voron builders did it? Did you do that? Did you read your local code?

The Trident and Voron 0.1 moving bed is a shock hazard, I don't think it would pass your local code even if grounded, because if the grounding wire breaks first and the mains second, it will still kill you. With a moving bed you would likely either need a 
https://en.wikipedia.org/wiki/Residual-current_device, some kind of automatic testing of the ground wire or a complete insulation of all metal parts.  Voron 2.4 may pass because the wire does not move.

The plastic sheet of the electronics compartment shall likely be a fire retardant kind.

bubnikv, 2022-02-20:
> The hall effect XY sensor requires 5V, however it outputs 3.3V. Formbot provided cable powers the board from 3.3V. Please document that the builder may need to split the cable provided by some kit providers to power from 5V, not 3.3V

BTW I just found out that the issue is not the Formbot kit, but the Octopus board documentation, it seems the documentation is outdated. I have created them an issue.
https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/105

That was a costly experiment, as the Formbot cable uses unusual color coding for the cables, thus I connected 5V to the hall effect outputs and fried the chip on the hall effect board. I was lucky to have the fried part on some other board, so I was able to fix the hall effect board quickly.