VoronDesign/Voron-2 #313: Q: Why did you decide for the gantry to move on Z axis?

Kreyren, 2022-02-20:
I wasn't able to find the answer in the repo so i am asking here:

Why did you decide for the gantry to move on a Z-Axis over mounting it to the top of the chassis and move the bed on Z axis?

Doing quick simulations it seems that this should cause an inconsistent printing at 800 (+-100) mm/s due to the momentum of the gantry?

or like is handling of this situation through firmware better in comparison to the mounting it to the top and move bed on Z axis?

Kreyren, 2022-02-20:
Or like is this projected to have a CNC rotor or printing at the angle in the future to make this design decision now?

BeardedTinker, 2022-02-20:
@Kreyren  - Voron 2 didn't change anything in terms of static bed - you are probably mistaken about versioning - V2 is not version but model 2 - you still have option to got for Voron Trident (previously 1.8).
Remember that number doesn't indicate version (as in newer model) but model number.
Also, by having static bed, lower centre of gravity, helps with stability and as a matter of fact allows higher speeds with this.

Kreyren, 2022-02-20:
@BeardedTinker does it still enable higher speeds assuming that the gantry mounted at the top with moving bed has a lower center of gravity (e.g. by adding a huge weight to the bottom of the printer)?

BeardedTinker, 2022-02-20:
Sorry, don't follow you - Moving bed (Z) is higher and as such has higher centre of gravity, while, fixed bed at the bottom has lower.. So not sure what you were asking? In terms of acceleration - I suggest that you join the Discord server as you will get more detailed and proper answer as not everyone is here or checks the GitHub.
But I have no problem with 10k accelerations on my V2, for printing, I go for quality so speed is not that important for me.

Kreyren, 2022-02-20:
> Sorry, don't follow you - Moving bed (Z) is higher and as such has higher centre of gravity, while, fixed bed at the bottom has lower..  -- @BeardedTinker (https://github.com/VoronDesign/Voron-2/issues/313#issuecomment-1046298972)

You said that it's about a center of the mass that enables it to be faster.

So assuming that both solutions have a weight added to the bottom of the printer to keep the center of the mass low -> Is the moving gantry still better?

> I suggest that you join the Discord server as you will get more detailed and proper answer as not everyone is here or checks the GitHub. -- @BeardedTinker (https://github.com/VoronDesign/Voron-2/issues/313#issuecomment-1046298972)

I don't use discord, but noted.

> But I have no problem with 10k accelerations on my V2, for printing, I go for quality so speed is not that important for me. -- @BeardedTinker (https://github.com/VoronDesign/Voron-2/issues/313#issuecomment-1046298972)

10K of what? m/s2 ?

BeardedTinker, 2022-02-20:
You can find manual and CAD files here and also you can go to https://docs.vorondesign.com/ for additional information.
Most of the answers to your questions are there. 
Also, for general questions and assistance Discord server is preferred way to go.

closed 2022-03-16 by raymondh2