VoronDesign/Voron-2 #322: Rear centre is about 6mm too big.

Macmagoo, 2022-02-26:
The Rear Centre Skirt is about 6mm too big it does not fit Voron 300.

stvptrsn, 2022-02-27:
> The Rear Centre Skirt is about 4mm too big it does not fit Voron 300.

Hi Macmagoo, can you provide any pictures?  I've printed an installed the parts on a 300 build so I'd like to see if something else might be going on.

Steve

Macmagoo, 2022-02-27:
Hi Steve, I hope the attached pictures help. It is my first Voron so perhaps it is something I have done. I am going to reprint it at some point today with a 4 mm reduction in length.

Thanks to the Voron Team for the work you all put into the design.

Regards

Andy

stvptrsn, 2022-02-27:
> Hi Steve, I hope the attached pictures help. It is my first Voron so perhaps it is something I have done. I am going to reprint it at some point today with a 4 mm reduction in length. Thanks to the Voron Team for the work you all put into the design. Regards Andy
> […](#)

The pictures do not appear to have come across. Are you on the Voron discord?  You can DM me there.  I doubt you need to re-print.  How do the side pieces line up?  Do you have the belt guards installed and are they pressed up against the Z drive modules? 

Steve

Macmagoo, 2022-02-27:
[image]
[image]

Macmagoo, 2022-02-27:
I have attached them here.

closed 2022-02-27 by Macmagoo

stvptrsn, 2022-02-27:
You have mismatched parts there.  The side opposite the power inlet is the old version or a front piece.  You need the keystone piece.

Steve

stvptrsn, 2022-02-27:
[image]