VoronDesign/Voron-2 #331: Wrong names of stl joints

Muszek1996, 2022-03-15:
There's name of MGN12 in .stl files but they are still mgn9
xy_joint_right_upper_MGN12.stl
xy_joint_right_lower_MGN12.stl
xy_joint_left_upper_MGN12.stl
xy_joint_left_lower_MGN12.stl

raymondh2, 2022-03-15:
The MGN12 denotes that those joints are for MGN12 on the X axis. Voron 2 doesn't support MGN12 on any other axis

SMTDesigns, 2022-03-16:
just a further note on this. 2.4r1 uses the mgn9 for the x axis and 2.4r2 release changed this recently to mgn12 (all the others are still the mgn9). I just printed the r2 parts and they are definitely mgn12.

stvptrsn, 2022-03-16:
I think the confusion was just in which axis the MGN12 referred to.  @Muszek1996 Now that that's cleared up I'm going to close this issue but if you need further clarification or anything let me know.

closed 2022-03-16 by stvptrsn