VoronDesign/Voron-2 #332: 2.4r2 PSU_stabilizer_50mm.stl

locki-cz, 2022-03-18:
PSU_stabilizer_50mm.stl
[image]

This stl is wrong and doesnt fit. In Fusion model is correct. 

[image]
[PSU_stabilizer_50mm correct.stl.zip](https://github.com/VoronDesign/Voron-2/files/8304141/PSU_stabilizer_50mm.correct.stl.zip)

SMTDesigns, 2022-03-21:
of note, the file above is not a zip file. Remove .zip from name to use.

locki-cz, 2022-03-21:
rename and remove .zip because .stl files arent allowed to upload

locki-cz, 2022-03-22:
> What is that M5 screw screwed into?

M5 T-nut

bofh69, 2022-03-31:
> M5 T-nut

That's not described in the manual

PR #343 (2022-04-08): manual update
page 98 Pulley -> Idler
page 100 Pulley -> Idler
page 197 added missing t-nut
page 220 split pwr in, added note about bed out
page 242 added side fan support heatset
page 270 added part highlight
page 272 added part highlight

closes #340, closes #338, closes #335, closes #332, closes #327
closed 2022-04-08 by raymondh2