VoronDesign/Voron-2 #333: M3 Theaded inserts are missing dimensions

SMTDesigns, 2022-03-20:
The BOM states just a M3 threaded insert with no details and the r2 assembly manual on page 31 mentions dimensions in the BOM (which aren't there). There are so many sizes of m3 inserts that it can be confusing to people that have not used them before.

Suggest adding dimensions to BOM as indicated "M3 4.6mm diameter and 4mm to 5mm long"

chri-kai-in, 2022-03-27:
Good initiative ! Aside that, i see it difficult to get other heat inserts than those requiring 4mm bore holes on the market. This might by a local issue, but all suppliers capable of delivery in the "big A" marketspace provide heat inserts with 3.9 or 4 mm bore hole reccommendations. It is hardly possible to get sufficient grip with those in todays 4.7mm bore holes in all Voron parts. On top to manual improvements, it would be maybe even more helpful to adjust the holes diameter in voron parts to the majority of available heat inserts on the market. On top, i still see it feasible to melt inserts for 4.7mm holes into 4mm bore holes. Worst case it would require prior drilling with a 4.5mm drill bit. Other way round i found it difficult...

SMTDesigns, 2022-03-27:
I produce 3d printed rocketry bay parts. I have used thermal inserts for years and it is important that the physical hole is sized for the insert for strength. What may be the best route is to look at what is available now and move all inserts to the same physical size. I would not recommend setting a larger insert into a small hole. Too much plastic gets melted and literally pushed out of the way.  Not all parts could use a larger diameter insert without making the surrounding plastic larger for support of the insert.

nemgrea, 2022-12-13:
fixed

closed 2022-12-13 by nemgrea