VoronDesign/Voron-2 #340: 2.4r2 manual power supply confusion

LoganFraser, 2022-03-26:
The manual on page220 has 3 sets of wires connecting to the inputs on the octopus, and appearing to be spliced together. This in not best practice, as the 3 output terminals on the power supply are not recommended to have the entire load on a single terminal screw pair. Please update the drawing with the following, that removes the redundant wiring to BED-Power and shows direct connections to the Mot-Power-in and Power-In.  
[image]

julianjc84, 2022-03-30:
This is how I wired mine too, My kit came with larger gauges of wire that was used to connect to the motors power tab too.

julianjc84, 2022-03-30:
I also found the orientation of the SSR on page 213 and on page 211 confusing.
page 213 shows load side and active wiring. on NO2
page 211 shows the Active to bed Reversed on NO1
[image]
[image]
.

closed 2022-04-08 by raymondh2

LoganFraser, 2022-04-24:
This isn't fixed though?  The diagram still shows junctioned wires instead of parallel ones and connection to BED POWER when BED OUT isn't used in the wiring diagram.

PR #343 (2022-04-08): manual update
page 98 Pulley -> Idler
page 100 Pulley -> Idler
page 197 added missing t-nut
page 220 split pwr in, added note about bed out
page 242 added side fan support heatset
page 270 added part highlight
page 272 added part highlight

closes #340, closes #338, closes #335, closes #332, closes #327