VoronDesign/Voron-2 #344: A report of the abusive languages on discord

Quickuser75, 2022-04-10:
I want to report the amount of abuse going on at the Voron discord and as the administration that are most should goven have not handled the complaint 
Fair and have not removed the member that use abusive language directed at members has been filed with the discord complaint team has been sent
I am hoping that this discord server be looked  into and the chat scrips from the brexited channel looked i to and i am hoping ever all member thay have used abusive language past or present be deleted and banned or the server be deleted and banned to the this kind as treatment is unacceptable in this day and age
Even by administration discord has access to all transcripts

flub9, 2022-04-10:
For context: I get email notifications for all V2 issues/updates so I can keep in touch with what’s going on.

Speaking fully professionally, this issue has some issues.
1) Not the right place, this is the repo for the V2 design, not if you feel insulted by discord. This is incredibly off topic for the repo.
2) I’m assuming this was written hastily or translated, because it is quite difficult to actually understand what the problem is
3) Based on your description, it seems you already resolved any issues by contacting Discord support
4) As a general git-hub rule, keep issues wherever they originated, so in this case, if your issue is in discord, keep it there.

Requesting to close.
-Flub

Quickuser75, 2022-04-10:
No it has not been closed until discord has finished the investigation

VinnyCordeiro, 2022-04-10:
> No it has not been closed until discord has finished the investigation this issue also show how unprofessional your design team really are

Strange, because no one on the dev team is paid for what they do. This is an entirely voluntary project, so you can't really talk about people being unprofessional.

Other than that, if your problem is on Discord, solve it on Discord. Polluting Github with complaints, even if you are right, won't help you.

stvptrsn, 2022-04-10:
This won't be productive.  This is not the correct channel for this.

closed 2022-04-10 by stvptrsn

Quickuser75, 2022-04-10:
No this is to show other uses how you really are and one of the abusive member stated he was a on the voron design team