VoronDesign/Voron-2 #345: Using unsafe and untested parts on BOM

Quickuser75, 2022-04-10:
I MAKE A SERIOUS WORING TO PEOPLE THAT BUILD
THIS PRINTER

I have been over your list of BOM and a lot of your pairs listed and unsafe and don't hold any genuine safety certificate 
Anyone that follow your list and be seriously accident and even death, one of your mains parts have not safety certificates

I have ordered all the parts use have listed and done real safety checks and have failed even an earth leakage test and main 
I HAVE ALSO SEE A POWERSUPPLU THAT IS LISTED BURST INTO FLAMES AND SENDING 250V down the low voltage lines
You have picked most aliexpress parts and serious unsafe 
This poses a serious health and safety concern 

I warn anyone that builds and uses the BOM that the amateur  voron designer have listed 

I would for a consumer election company and do emi and safety tests on electronic even hobbies based 
And this printer would as built would fall the test

Maybe the set files would be safe upto a point
I am looking it to this also

They offer a serial where you can shoe your printer working and the wiring, but that is not a emi or safety test

SMTDesigns, 2022-04-10:
Please pass on the test results to the community here, so we can look at what you have done. Also, point out what parts specifically do not have certificates rather than just sending a general list of negative comments. The screen shot you made would be helpful also. Thank you.

Quickuser75, 2022-04-11:
Oh looks a rocket designer to vorons rescue, do you use a voron by any chance... maybe you should do the tests close my case

bdbell, 2022-04-11:
Actually, it's a really common part of the accepted scientific method to ask for someone else's test results as a starting point for proving or disproving something.  He is one of the designers and is openly looking for the evidence you have in order to start his investigation. If you have legitimate test results then please provide them as it would be helpful.

DrDMo, 2022-04-11:
Dude, just leave.

Quickuser75, 2022-04-11:
I don't have any of them I have sent them to the UK British standards to be investagated

bdbell, 2022-04-11:
> I don't have any of them I have sent them to the British standards in the UK to be investagated unlike the US we have powered to have item imported investagated for safety and the cases banned from entering the country 

I look forward to seeing the results.

Quickuser75, 2022-04-11:
Github is unsafe to post personal information 
As of this I will not be sending anything

I am posting to mind other that using aliexpress is unsafe very small procent of electronic are tested and just has a shiny sticker them can be get from ebay

Quickuser75, 2022-04-11:
I will not be responding again

Quickuser75, 2022-04-11:
Sorry SMTDesigns, when are you going to Make a better 3D Printer

closed 2022-04-11 by Quickuser75

DrDMo, 2022-04-11:
What an idiot.

shiftingtech, 2022-04-11:
I'm locking this, just to avoid any more aggression on any side