VoronDesign/Voron-2 #348: Minor manual issues

gjsmo, 2022-04-20:
I am about halfway through my first build and I'd like to make some comments on the manual:

* Overall, printed parts are not labeled as to what their filename is. This can make determining which part is which difficult - it would be nice if these had labeling. A mark or other indication as to color (main or accent) would be very helpful as well.
* Page 20 "Belt Drive Shaft Assembly" - at least for the bearings and shafts I chose, bearings were press-fit. They also do not have dimensions listed. The best way I found to do this was to use a vice and a deep socket to push on the inner races, then when the bearing gets close to the final location move it by a millimeter or two and test fit. Dimensions and better instructions, possibly a printed press tool, would be helpful here.
* Page 23 "Z Drive Motor Mounts" - the motor mount no longer looks like this.
* Page 47 "Dual Rail Install" - the rail installation guide isn't usable for the the second rail to be installed, as it interferes with the first rail. Possibly this part could be updated with a shorter "leg" on one side? However, it appears it's beneficial to leave these loose as during carriage alignment they will probably bind.
* Pages 52 "Y Rail Install" and 53 "Front Idlers" - these would make much more sense if they were reversed, as page 52 references the idlers yet to be installed on page 53.

DunarDolorgiet, 2022-04-20:
Can you please check if you are using the current version of the manual? 
First page bottom right should read: "VERSION 2022-04-09".

julianschill, 2022-04-20:
I think you are referring to the old manual, becaude most of this has been adressed with the 2.4r2 manual.
Regarding your second point: This shouldn't be press fit. You probably sourced the wrong shafts.

gjsmo, 2022-04-20:
You're both correct, this is the 2.4r1 manual ("2021.01.20") as well as 2.4r1 parts. I sourced 90% of everything last year, before 2.4r2 came out - figured I might as well finish it! I'm glad to see the manual has had a significant update, it does seem much clearer and well laid out now. Labeling of part filenames would still be helpful (it's nice to load up the STL sometimes to assist in locating the part), although I do appreciate the labeling of accent colored parts.

As far as the shafts go, there isn't an indication of which shafts to use apart from "5mm". If they shouldn't be press fit this should be clearer. As I'm in the US, a significant portion of the hardware is from McMaster - far faster than AliExpress. The linked shafts in the BOM do not have accurate tolerances or dimensions (at least not in English), so it's hard to tell what fit class is appropriate. I ended up with McMaster 1265K44 which are precision ground to 5mm +0.000/-0.005mm. Along with common 625-2RS bearings this is a press fit. To be clear, this isn't a problem functionality-wise, just assembly-wise.

There's a separate sourcing issue regarding US vs. European "2020" extrusion - again, it isn't entirely clear that there's a difference between the two, and the fastest to source in the US is from 80/20 (when I purchased the frame, Misumi was quoting something like 26 weeks lead time!). Unfortunately many components expect a 6mm gap rather than a 5mm gap so I've just been modifying those with a hobby knife. I figured I'd mention it on the topic of sourcing, I can make another issue if that would be more appropriate.

raymondh2, 2022-04-24:
Closing as the original issue is resolved

closed 2022-04-24 by raymondh2