VoronDesign/Voron-2 #360: mini12864_case_front fillet edge

whatthefilament, 2022-06-02:
When printing a new front case for my V2.4 I noticed that the heat-set inserts are designed to nestle in the front case body when inserting the accent piece. However, the cutouts for the heat-set inserts began halfway down the face making it very difficult to install the accent piece. Simply extending the cutouts up the face made the assembly and fitment much better. 

Wasn't sure how to upload the modified files so here is a link to a repository for them. Please let me know if there is a better way. 

https://github.com/whatthefilament/Voron-2.4-Mods/tree/main/Display%20Panel

[image]

stvptrsn, 2022-06-06:
The cutout for the heatset inserts have no lips to catch on anything when inserting the accent piece into the main housing so I do not see the need to extend those holes to the front surface.  I would check for print artifacts on your pieces that may be causing you difficulty.

Section analysis showing there are no surfaces that would prevent the accent piece from being inserted into the front piece.
[image]

closed 2022-06-06 by stvptrsn

whatthefilament, 2022-06-06:
I get what you're saying. 

You can see the cutout fits the insert perfectly. However, when installing the front plate after the insert is installed it has a very hard time getting passed that edge. If that edge is gone. The heat-set insert slides in easily  

[image]

stvptrsn, 2022-06-06:
Well there's the problem.  You insert the accent piece into the housing THEN insert the heatset while holding those two parts together.

whatthefilament, 2022-06-06:
Looks like my guide was corrupted. Which could have been part of the problem too. 

[image]