VoronDesign/Voron-2 #365: Multiple discrepancies between STL and CAD models of electronics bay brackets

ReXT3D, 2022-06-14:
While assembling my 2.4v1 I was not thrilled with how loose the electronics bay components were on the DIN rails and I printed a number of upgraded brackets and mounts from 2.4v2 release STLs.  I then decided to make some additional changes to the parts and downloaded the F360 model to modify.  To my surprise, I immediately noticed differences in the models.  The items below are just two that I immediately identified and there could be more discrepancies.

1. **PSU_stabilizer_50mm.stl** does not have slotted hole for the PSU screw and will not fit without pushing up the decking material into the chamber.  The CAD model on the other hand has the correct slotted hole.  I also noticed on Discord that some people were posting photos of their PIF part kits that were printed with the STL model that has fitment issues.

[image]

2. **pcb_din_clip_x3.stl** is significantly different dimensionally from the CAD model.  In this case, my personal preference is actually for the STL model as it provides a much more secure mounting on the DIN rails while the CAD model clip is somewhat loose, especially when used alone such as on the 5V PSU.

[image]

[image]

DouglasJOlson, 2022-06-30:
I can attest the PSU Stabilizer 50mm STL as posted in the r2 STLs does not fit the recommended Meanwell PSU, the single Screw hole is off by 3-4mm, the adjustable screw hole in the CAD model works fine.  The STL should be updated with the CAD version to prevent further issues.

Jack-Lenders, 2022-10-10:
The STL for the Meanwell PSU does not align with the actual screw hole on the PSU. I think it is based on the height of the older DIN rail mounts.

[image]

DouglasJOlson, 2022-10-10:
> The STL for the Meanwell PSU does not align with the actual screw hole on the PSU. I think it is based on the height of the older DIN rail mounts.
> 
The bracket in the CAD is correct, it uses a slot for the screw position, so can work with a wider variety of psu arrangements.
It needs to be updated in the downloadable STLs but as yet, this has not been done.

Jack-Lenders, 2022-10-10:
Posted this to bump your correct findings and maybe so the PIF team can update their plates and provide the correct bracket to new builds.

jsaintrocc, 2022-11-23:
Hi @DunarDolorgiet I've got a PR submitted with a fix for this. Could you help me get someone to review it? Happy to make any changes necessary and looking forward to contributing.

PR #405 (2023-03-02): Multiple discrepancies between stl and cad models of electronics bay brackets #365
fixes #365