VoronDesign/Voron-2 #366: Voron 2.4 Manual update and sourcing guide update

winstonwoff01, 2022-06-19:
Pages 227,229 and 232 on the Voron 2.4r2 VERSION 2022-02-19 reference flat head screws to attach the cable chains but the 2.4 Sourcing guide does not mention needing flat head screws.

nemgrea, 2022-07-25:
added

closed 2022-07-25 by nemgrea

GadgetAngel, 2022-07-25:
> added

The sourcing Guide spreadsheet has not been updated

GadgetAngel, 2022-07-25:
In fact there are discrepancy between the Spreadsheet and the Sourcing Guide.  Please check this over and fix

nemgrea, 2022-07-25:
i believe you are confusing the BOM and the souring guide, the sourcing guide has indeed been updated

GadgetAngel, 2022-07-25:
> i believe you are confusing the BOM and the souring guide, the sourcing guide has indeed been updated

You are correct the Sourcing Guide was updated, but now the BOM is not showing all the parts.  Who updates the BOM?  Please let them know the BOM is not in synch with the sourcing guide.  M3x6 FHCS and M3x10 FHCS are both missing from the BOM!

nemgrea, 2022-07-25:
bom updates are handled through a seperate private repo, 

PR has been submitted for approval that includes the following:
- added m3x6FHCS per PR#366 on V2 repo
- fixed formatting for m3x10 FHCS as they were not displaying

GadgetAngel, 2022-07-25:
> bom updates are handled through a seperate private repo,
> 
> PR has been submitted for approval that includes the following:
> 
> * added m3x6FHCS per PR#366 on V2 repo
> * fixed formatting for m3x10 FHCS as they were not displaying

Thank you, I will remember that info