VoronDesign/Voron-2 #367: btt mini 12864 connection issue- the leds on but nothing on the screen

mitofun, 2022-06-21:
Hi FOLKS how are you
could you please help me- cannot connect the btt mini 12864 to the skr 1.4t and raspberry pi 3 model A+
i follow the official instruction https://docs.vorondesign.com/build/electrical/mini12864_klipper_guide.html
have removed the R1 and C6 and uploaded the code to the printer.cfg - 

**[display]
##  mini12864 LCD Display
lcd_type: uc1701
cs_pin: P1.18
a0_pin: P1.19
encoder_pins: ^P3.25,^P3.26
click_pin: ^!P0.28
contrast: 63

[output_pin beeper]
pin: EXP1_1

[neopixel fysetc_mini12864]
pin: EXP1_6
chain_count: 3
color_order: RGB
initial_RED: 0.0
initial_GREEN: 0.2
initial_BLUE: 0.0

##  Set RGB values on boot up for each Neopixel. 
##  Index 1 = display, Index 2 and 3 = Knob
[delayed_gcode setdisplayneopixel]
initial_duration: 1
gcode:
        SET_LED LED=fysetc_mini12864 RED=0 GREEN=0 BLUE=1 INDEX=1 TRANSMIT=0
        SET_LED LED=fysetc_mini12864 RED=0 GREEN=1 BLUE=0 INDEX=2 TRANSMIT=0
        SET_LED LED=fysetc_mini12864 RED=0 GREEN=1 BLUE=0 INDEX=3** 

after restarting have got this message

**mcu 'mcu': Unable to encode: config_spi

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.10.0-278-g7c964e5f, mcu=v0.10.0-419-g4490a584, rpi=v0.10.0-278-g7c964e5f

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer**

if i comments the display section - the mcu starts but there is no any data on display only light on 

so if anyone could help me will be appreciated! 
thank you

mitofun, 2022-06-21:
[image]
[image]
[image]
[image]

mitofun, 2022-06-21:
have managed to start it! yep need to remove the R1 and C6 (the cables good no need to change anything!) 
[image]
all you need to upgrade the klipper and the moonraker! 
the code i have used as the following

#############################################
## BOARD PINS + DISPLAY + SD CARD
#############################################
######################################################################
# BTT SKR 1.4 Turbo 
######################################################################
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"
######################################################################
# Fysetc Mini 12864Panel v2.1 (with neopixel backlight leds)
######################################################################
[display]
lcd_type: uc1701
cs_pin: EXP1_3
a0_pin: EXP1_4
rst_pin: EXP1_5
contrast: 63
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
contrast: 63
## Some micro-controller boards may require an spi bus to be specified:
#spi_bus: spi
## Alternatively, some micro-controller boards may work with software spi:
#spi_software_miso_pin: EXP2_1
#spi_software_mosi_pin: EXP2_6
#spi_software_sclk_pin: EXP2_2
[output_pin beeper]
pin: EXP1_1
[neopixel fysetc_mini12864]
pin: EXP1_6
chain_count: 3
initial_RED: 0.2
initial_GREEN: 0.2
initial_BLUE: 0.0
#initial_RED: 0.4
#initial_GREEN: 0.4
#initial_BLUE: 0.4
color_order: RGB
# For Mainsail and Fluidd
#############################################
[display_status]
# for display messages in status panel
#############################################
[pause_resume]
recover_velocity: 50
# for pause/resume functionality. 
# Mainsail needs gcode macros for `PAUSE`, `RESUME` and `CANCEL_PRINT` to make the buttons work.
######################################################################
# Plug pin locations
######################################################################

# Some micro-controller boards and displays use inconsistent labeling
# for the EXP1 and EXP2 headers. The following diagram shows the
# correct location of pin 1 along with ground and power pins on the
# EXP1 and EXP2 plugs:
#
#          EXP1:                        EXP2:
#   +-----------------+          +-----------------+
#   |  o  o  o  o  5V |          |  o  o  o  o  o  |
#   |  1  o  o  o GND |          |  1  o  o  o GND |
#   +------     ------+          +------     ------+
#
# Some boards may have the cutout in the wrong location. If so, it may
# be possible to carefully pry the plastic header off of the pins with
# a small screwdriver, then correct the orientation and reapply the
# plastic header.
######################################################################
# SET DISPLAY NEOPIXEL
######################################################################
[delayed_gcode setdisplayneopixel]
initial_duration: 1
gcode:
#        SET_LED LED=fysetc_mini12864 RED=1 GREEN=1 BLUE=1 INDEX=1
#        SET_LED LED=fysetc_mini12864 RED=1 GREEN=0 BLUE=0 INDEX=2
#        SET_LED LED=fysetc_mini12864 RED=1 GREEN=0 BLUE=0 INDEX=3 TRANSMIT=0
        
        SET_LED LED=fysetc_mini12864 RED=0 GREEN=1 BLUE=0 INDEX=1 TRANSMIT=0
        SET_LED LED=fysetc_mini12864 RED=0 GREEN=1 BLUE=0 INDEX=2 TRANSMIT=0
        SET_LED LED=fysetc_mini12864 RED=1 GREEN=1 BLUE=0 INDEX=3

raymondh2, 2022-06-21:
Most of the people who will be able to help you are on our reddit and discord. I'd suggest going to the Vorondesign.com website and going there.

closed 2022-06-21 by raymondh2