VoronDesign/Voron-2 #376: Cable chain retainer brackets - how is this meant to work?

Alex9779, 2022-07-15:
I overhauled my Voron 2.4 with some of the 2.4 R2 parts because of some issues and a switch to the Stealthburner.
But I do not get how the cable chain retainers should work, mentioned on page 232 of the manual. I have to use those because I changed my A/B drive units too, the original parts I ordered from a print-it-forward turned out to be a little off.
I printed the chable chain retainer brackets to fix my generic three hole cable chain but the hole of the retainer is about 4mm in diameter, an M5 screw will sit tight but is too big for the cable chain holes.
The M3x10 FHCS of the manual just slides through and there is nothing it screws into so.
For the moment I used a hex nut, but I don't think this is the way it should be...

Is there something missing? Is the hole of the retainer too big?

DunarDolorgiet, 2022-07-17:
Have a look at page 231, the updated ones use heat set inserts instead of hexnuts.
[image]

Alex9779, 2022-07-17:
Ahhh cool, sorry I missed that... Thanks!

closed 2022-07-17 by Alex9779