VoronDesign/Voron-2 #378: Manual Specifies wrong wire thickness.

neeamradia, 2022-08-03:
Page 210 of the 2.4r2 manual specifies:

'Cables should be at least 1mm^2 (AWG18) or thicker depending on local regulations.'

18AWG = 0.75mm^2
0.1mm^2 = 16AWG?

Which one should it be? 18 or 16 AWG?

[image]

nemgrea, 2022-08-05:
American 18 gauge wire is listed at 0.0403in diameter which is above 1mm. where are you getting .75mm from?

closed 2022-08-05 by nemgrea

neeamradia, 2022-08-06:
> American 18 gauge wire is listed at 0.0403in diameter which is above 1mm. where are you getting .75mm from?
> 
> EDIT: i believe you may be looking at cross sectional area of which it would be closer to 0.75 when what is called out in the note is thickness (i.e. diameter)

Sorry - yes that is likely that I am looking at the wrong dimension.