VoronDesign/Voron-2 #380: Keystone panel is only for one size

TekuSP, 2022-08-17:
Keystone panel in R2 is only for one size.
https://github.com/VoronDesign/Voron-2/blob/Voron2.4/STLs/Skirts/keystone_panel.stl

Compare of front skirt 350mm and Keystone Panel:
[image]

raymondh2, 2022-08-17:
It's for all sizes. The rear skirts only have the center piece as different sizes now. This was to simplify unique part count and make it not matter what size printer you are building for PiF..

closed 2022-08-17 by raymondh2

TekuSP, 2022-08-17:
Oh yes you are right. I missed that somehow.
On R1 it was front_rear_skirt and here it is only front_skirt and rear middle part is different.

Thanks!