VoronDesign/Voron-2 #383: Cura profiles don't show up

narf222, 2022-08-30:
hi all,

i downloaded the voron definitions for cura from github (here) and placed the files in the respective folders. i was able to import the new printer to cura, BUT the printing profiles aren't visible.
i'm using cura 4.13.1 under win7 and was also able to verify the exact same thing on a virtual machine. 

am i missing something or is there an invalid parameter in the quality profile file (*.cfg) ? - i saw that they are from 2019. maybe something is not supported anymore.

thanks for all your good work. i'm loving the "voron experience" so far :) 
narf

raymondh2, 2022-09-06:
Support questions like this are unlikely to answered here. Please try our reddit or discord

closed 2022-09-06 by raymondh2

narf222, 2022-09-07:
sorry, but this is just a "support" question if i did something wrong. when the profile you offer is outdated/incompatible with the current cura version, then it's a bug report :) 
at the moment i'm not sure what it is - but then, it should be fairly simple to copy the files to the corresponding folders.