VoronDesign/Voron-2 #385: Feature Request for documentation versioning

Sonnenwerk, 2022-09-04:
As a user i would like to have not only the documentation date, but also the git commit number visible in the manual (pdf), so that i can look at the (last and or/future) comits to see, what has changed since my last document version.

I also would like to have the date (and/or) comit numer in the footer of every page, in case, i just print the docs partly, or reference them in discord or whatever.

claytono, 2022-09-06:
You can't know what the git commit SHA is going to be before committing the data, so you can't really put the git SHA in the manual.

Sonnenwerk, 2022-09-16:
Sadly i am not into details of github. With subversion i can input a variable "%version" into the document and while i update/upload my file into repository the variable is substituted by the real value of the comit. If this is not possible with github, you can close the FR.
cu

closed 2022-09-16 by raymondh2