VoronDesign/Voron-2 #388: [Manual] Mention that Hall Effect PCB is for legacy support and not in any way an upgrade

ChaosBlades, 2022-09-21:
In the manual the Hall Effect is presented as an upgrade. At least I interpreted it that way and it seems several people in the Amazon reviews for the Hall Effect PCB I bought thought this as well. Turns out the Hall Effect sensor is worse in basically every way from the micro switches as I found out asking about it in Discord. This should really be pointed out in the documentation or just removed from the manual all together.

raymondh2, 2022-09-22:
I'll bring it up to the team.

nemgrea, 2022-09-22:
> In the manual the Hall Effect is presented as an upgrade. At least I interpreted it that way and it seems several people in the Amazon reviews for the Hall Effect PCB I bought thought this as well. 

[image]

its shown as an option, i dont think there is anything in here that expresses that one is better than the other or that one would be an upgrade over the other, simply two options are shown.

ChaosBlades, 2022-09-22:
I don't think that is good enough. According to my discussion on Discord the hall effect isn't any more reliable than micro switches, it is sensitive to chamber temp, switching to PLA when tuned for ABS it could go so out of spec enough to cause a crash, it drifts over time, and you can't even increase home speed. It is not an improvement in any way.

What I was told was...

> StevenB V2.3370 — Yesterday at 2:55 AM
Because it was a test that failed and it's still kept as an option because of that.

When I go to some of these selling on Amazon I see people complaining that the Hall Effect PCBs are using a different connector (Molex instead of JST) so they need to re-cable when moving from Microswitch. Which is a downgrade. So clearly myself and other have fallen victim to this simply being show as an option. I am sure there is some kind of lesson here on assuming things.

nemgrea, 2022-09-22:
i agree is is not any more reliable at all NOR is it ever claimed to be more reliable, im not sure why you are assuming that it would be more reliable when it is not presented as "better" or of higher quality in any of the documentation. it is simply showing steps for an OPTION that exists and happens to be popular. no claims of its benefits or performance are shown stated or implied in the manual...

subatomicglue, 2022-09-23:
I wonder if OP is getting at:
 - Shouldn't the default of voron's manual be to clearly highlight the
"best", for the optimal voron experience?   Like, If I'm presented with 2
options, and I know not which is better, shouldn't there be something
helping guide me to choose, so that I make the best voron printer available
and not go down "less-good" paths?

raymondh2, 2022-09-23:
Our BOM already lists the normal microswitch setup and not any of the PCBs. But these PCBs are included in the manual because they are so common and we get too many questions when we don't include the instructions for it.

asdasdwa-v02679, 2023-05-30:
Could we possibly solve this problem by adding text on page 191 below the images that has a warning specifying that although the pcbs can simplify wiring they tend to introduce their own issues with repeatability and aren't recommended.  

or is it just the hall effect pcb? if so perhaps it would read more like "although the hall effect sensor pcbs have become super common and can simplify wiring, they tend to introduce their own issues with repeatability and aren't recommended."

does this seem like a good solution according to yall? if so and people approve I am happy to make the edit to the page :)

asdasdwa-v02679, 2023-06-10:
@majarspeed  I understand that but It seems that Nemgrea marked it as manual so I was trying to find a solution. also It does seem a fair few people get confused by this page and so a edit might be in order ?