VoronDesign/Voron-2 #390: Manual for Z axis missing Z idler assembly + suggestion for manual.

Eclypse3D, 2022-09-28:
In the manual while going though the parts, on page 109 for Z axis it shows Z idler at the top of the gantry, it then again references the Z idler on page 117 just before the belt install, but nowhere is the page on the assembly of the Z idler.  No mention what or how many bolts, nuts, or anything hardware related is mentioned, it only mentions extending the idler during the belt install.  Needs update and clarity.

Suggestion for the manual over all. On each section of the build just after the cad drawing, on the overview page, it would be prudent to include the parts list for each section.  i.e. z idler x4, M3x20 x8, M3 nut x1 etc.  For each section.  This would help the builders consolidate parts in a compartmentalized fashion.

julianschill, 2022-09-28:
Z Idler assembly is on page 48 - 50.

Not sure about the part list. On the one side it would be nice, on the other side it would be a lot of work to do and is more like a "nice to have" than a necessity. And it would be quite a lot of work to prepare.

Eclypse3D, 2022-09-28:
Oh jeese how did I blast past those pages, I think maybe the browser skipped over it while scrolling I was going through the parts I had printed to inventory from the manual and never saw those pages, well nvm then.... 

As far as the manual thing, how much harder would it be? the time was taken to make the manual, and create a bom.  I am not sure that is a is a stretch considering the work already done.  If anything, it seems like it was more of an oversight as most other  manuals usually give you some sort of parts list.  Yes we have a bom, and technically that with the manual would be a rough equivalent, but its not a replacement.  Yeah it would be nice to see at least future manuals with it during creation.

Eclypse3D, 2022-09-28:
Pages found, false alarm...

closed 2022-09-28 by Eclypse3D