VoronDesign/Voron-2 #395: BTT Manta MP4 and MP8 boards not supported

Krisstank, 2022-10-08:
can we get Controller mounts created for the 

BigtreeTech
Manta MP4 and MP8 created

nemgrea, 2022-10-10:
CAD if freely available for anyone to submit stls for new controller mounts, however a github issue is not the appropriate place for requests like this. github issues are for identifying and fixing issues with existing components or documentation not for requesting work from community members

closed 2022-10-10 by nemgrea

valadas, 2022-10-10:
@nemgrea I may have a few things to submit soon, just out of curiosity, what is the preferred process:
1. Just the stl in the PR, named so it makes sense?
2.  A new Step and f3d file with the updated options (or not because only the dev team touches those) ?

nemgrea, 2022-10-10:
if its just mounting solutions id probably just submit them to the usermods repo, i think in order to add support for a new board into main we'd at least want to start with adding a relevant config file. and then work on how best to integrate a new solid file to the main assembly but thats probably more of a group dev decision. 

but yea id guess if you have a config file an stl and a step file then the best course would be to PR the .cfg and the .stl into their respective folders and then download the assembly .step and .f3d and insert your step file into them and re-upload.